FreeJoy-Team/FreeJoyConfigurator #2: Can't save

mahdiyari, 2019-12-23:
Hi
Basically it doesn't work
I can't save or load from device (it says connected but none works)
I'm on windows 10

vostrenkov, 2019-12-23:
What says log block in the bottom?

mahdiyari, 2019-12-23:
nothing
stuck at Requesting config..

vostrenkov, 2019-12-23:
Did you flash to device hex file from releases or compiled it yourself?

mahdiyari, 2019-12-23:
I just used your file from releases

vostrenkov, 2019-12-23:
Give me some time to make clean install and check this

mahdiyari, 2019-12-23:
do I need to use the "Firmware Programmer" inside app?

vostrenkov, 2019-12-23:
nope, it is bootloader feature to flash firmware without programmer

mahdiyari, 2019-12-23:
I will test with another board that I have (this one has different reset button)

vostrenkov, 2019-12-23:
ok, report please your results

mahdiyari, 2019-12-23:
The first board displays 128kb flash size in stm32 st-link utility
the second board 64kb flash size

The app seems to work with the second board but still has problems
get config works well I think
If I change buttons and write config, I get "Config written" message but after pressing get config, all buttons revert back to the default settings

mahdiyari, 2019-12-23:
now it works with the second board
I will appreciate if you fix the problem for the 128kb flash size model too

can I use rotary encoder as joystick with freejoy? I added and it acts like a button

vostrenkov, 2019-12-23:
I will try to fix 128kB problem in next release. 
You can use encoders as buttons, rotating to one side triggers one button, to other side - another button

mahdiyari, 2019-12-23:
Can you tell me how can I edit and build freejoy myself?

vostrenkov, 2019-12-23:
you can use Keil uVision IDE or another option is to build with Makefile with gcc compiler

mahdiyari, 2019-12-23:
Thank you
what about FreeJoyConfigurator?

vostrenkov, 2019-12-23:
It is Visual Studio 17 WPF application project

closed 2020-03-31 by vostrenkov