FreeJoy-Team/FreeJoyConfigurator #5: The shift register LATCH/DATA pin is confusing

jieweiyang, 2020-03-07:
In Configuration interface, the LATCH/DATA pin configuration cannot tell which register is related, when there are multiple Shift register group connected.

vostrenkov, 2020-03-07:
Data/Latch pins for several shift registers groups are mapped sequentialy (pins count starts at A0 and ends at C15). 
For example:
B0 -> Latch
B1 -> Data
C13 -> Data
C14 -> Latch

In this case B0 and B1 are related to first shift register and C13, C14 to second shift register.
I think i can make Latch pin common for all shift registers group (like SCK), i will do this for the next release

jieweiyang, 2020-03-07:
Understand now, but hopefully this can be proper describe in the configurator.

closed 2020-08-30 by vostrenkov