FreeJoy-Team/FreeJoyConfigurator #6: How do I run the FreeJoyConfigurator?

tistructor, 2020-03-31:
Ciao,
What application do you need to run FreeJoyConfigurator?

vostrenkov, 2020-03-31:
Hello,
You only need .NET version equal or higher 4.5.1.
Download .zip archive from realizes and run installer

tistructor, 2020-03-31:
Thanks,
Done.
Great project, I tested the bluepill card and it works great.

freakerman21, 2020-04-04:
I can't see any installer on that zip

vostrenkov, 2020-04-04:
Please download .zip archive from releases page: https://github.com/FreeJoy-Team/FreeJoyConfigurator/releases
[image]

303HOLY, 2020-05-12:
From some reason I can not start configurator. Is it anything what Im doing wrong
[image]
[image]

vostrenkov, 2020-05-12:
Do you run installer from archive without unpacking? If yes please unpack it before installing

303HOLY, 2020-05-12:
Thanx for reply.
Yes of course first was unzip

303HOLY, 2020-05-12:
I unpacked this zip to desktop then when I dubble click setup.exe or when I run it as administrator I have poup this window as I attached

vostrenkov, 2020-05-12:
Please try these solutions: https://answers.microsoft.com/en-us/windows/forum/all/your-administrator-has-blocked-this-application/acfb1b75-063b-46f0-a176-06ab7798c9b3
I'd start from turning off UAC. If it wont help then installing in safe mode

303HOLY, 2020-05-12:
I've checked two option - still this ugly window pop up :(

303HOLY, 2020-05-12:
do I have some virus or something - this is very strange :(

303HOLY, 2020-05-12:
Of course any other programs run just perfect

vostrenkov, 2020-05-12:
I dont think the reason is virus. Quite strange behaviour though. 
Looks like your system dont allow to install not signed software for some reason

vostrenkov, 2020-05-12:
Have you tried to start it from system explorer but not unreal commander?

303HOLY, 2020-05-12:
...Yes I have it in unpacked in the folder on my desktop now

303HOLY, 2020-05-12:
Probably Ive set something in somewhere in security settings and I have now idea what.
Or somewhere in disck settings

303HOLY, 2020-05-12:
But funny things is that everything else is working just perfect

303HOLY, 2020-05-12:
Anyway please HELP :)

303HOLY, 2020-05-12:
Or like I said maybe I just recived some "ugly unwanted person" somwhere in my PC who dont like your beautyful program

303HOLY, 2020-05-12:
is it normal that in security is not checked this
[image]

303HOLY, 2020-05-12:
this special permitions

vostrenkov, 2020-05-12:
Running in safe mode usually resolve all permission problems. If it doesn't work for you i can advice you to build it yourself from sources with visual studio

303HOLY, 2020-05-12:
first Ill try in safe mode one more time

303HOLY, 2020-05-15:
Hello there.
After instaled NEW system everything working just perfect.
Thank you very much for your help but thousands times MORE for this beautiful program.
Regards HOLY
BTW
Is it a way to put longer USB name for my product?

vostrenkov, 2020-05-15:
Hi, i'm glad to hear you managed to solve your troubles.
USB name lenght is limited by 20 characters. Do you need more?

303HOLY, 2020-05-15:
Thx for replay. 
Is it possible to put 30 characters?

303HOLY, 2020-05-15:
like that for example:
SimPit303 P-51 Engine Panel

303HOLY, 2020-05-15:
Ive notice that from some reason windows do not see all my name but its going to put shorter
is something Im doing wrong?

vostrenkov, 2020-05-15:
It is not possible in current version of the software. 
But i'm working on new version right now and it will have more memory space allocated for configuration data. So i can extend max USB name lenght to 30 characters

303HOLY, 2020-05-15:
in dynamic hid config should I always every time change the PID numer to see the diferent name of my device in windows?

303HOLY, 2020-05-15:
...when I decide for example to chage the name of device

303HOLY, 2020-05-15:
Vostrenkov are you on Discord : HOTAS/HOSAS/SIMPIT maybe?

vostrenkov, 2020-05-15:
This is Windows problem, it remember device name at it first connection and does not change it even if it changed in the descriptor.
You have to delete the information about your device from Windows registry to let it change the name.
If you change PID windows just see "new" device and remember its name at first plug

vostrenkov, 2020-05-15:
Yes, there are subchannel freejoy in that channel. You can find me there as Magners

closed 2020-06-16 by vostrenkov