FreeJoy-Team/FreeJoyConfigurator #11: Trouble reading/writing config

Gaskrank, 2020-04-24:
Controller is "connected" in the configurator and is running with standard config but I can't seem to read/write from/to the controller from the FreeJoyConfigurator. Tried on two different PCs already. What am I doing wrong? I want to modify the config for a pedals controller.

vostrenkov, 2020-04-24:
which version of firmware and config tool is used?

Gaskrank, 2020-04-24:
I tried both firmware/configurator 1.4.1 and 1.5.1 - same result.

vostrenkov, 2020-04-24:
Sometimes (very seldom, I only got it twice) HID library may block reading or writing to HID device. In this case reboot of the PC is required.
USB cable quality and its length may make sense. Please try connecting with short USB cable. 
Also try to connect to USB port placed on the back of the PC, front panel ports may not work properly.

Gaskrank, 2020-04-25:
I tried five different USB-cables now and with the fifth it is working every now and then on my PC now. On the laptop it still won't work for me. But at least I could write the config to the controller. Thanks!

vostrenkov, 2020-04-25:
It is quite strange, so I guess there are some problems with the hardware (bluepill board). 
Usually you don't have to try all your cables to make it work. If you are going make another device in the future you'd better order bluepill from other supplier

closed 2020-06-16 by vostrenkov