FreeJoy-Team/FreeJoyConfigurator #16: Three position toggles, ON-OFF-ON

snakepl88, 2020-06-23:
Awesome project, thank you for the hard work. 
I have a couple 3 position toggles I want to use (like the boat switch or pinky in a A-10C throttle). Can I map a button when it is in OFF state and how would I do that? I can do it with 2 position toggles, but 3 positions I can't figure it out.

vostrenkov, 2020-06-24:
Hello!
Technically On-Off-On toggle switch is 2 switches with common terminal, so to handle "off" state you need to be able to map 2 physical buttons to 1 logical button with some relationship logic (log_state = !phys_state1 && !phys_state2). It is not possible in FreeJoy yet (i dont know where it possible without scripting)

snakepl88, 2020-06-24:
Just to follow up, I was able to map the OFF states to 2 logical buttons and made those OFF states momentary to avoid conflict. It seems to work fine in DCS for now. Scripting support would be great but I do understand it's not a priority. Thanks again for quick reply.

vostrenkov, 2020-06-24:
Yes, not-latching buttons is a kind of a solution but it is not "honest" on-off-on handling. Scripting is not a priority as it is possible to do this by some mappers.
I'm glad your issue is resolved, stay in touch in case of future troubles

closed 2020-06-24 by vostrenkov