FreeJoy-Team/FreeJoyConfigurator #17: Axis to 16 Buttons would be nice to encode a 16 Button resistance matrix

bayerla, 2020-07-11:
Hi from Munich.
I just started, flashing went well. 
I would like to add a  [https://github.com/RobotDynOfficial/Documentation/wiki/Button-Keypad-4x4-module-Assembled]

It would be nice do add 16 Positions to the Axis to Buttons. May be a problem to recognice 16 values?

[image]

Then something else you: I use a Robody Black Pill and documented the differance to flash.
[image]
May be any problems usind Black Pills ?? ( else than 4 Pins less and other Position)
If not you can add this pikture to wiki for others.

May be any problem by UART flasching using 5V Pin from the UART instead of extra USB connection ?

vostrenkov, 2020-07-11:
Hello,
16 buttons require some code changes, though it is possible. I can not promise you i will add it soon but i added this to the task list.

BluePill and BlackPill are pretty the same in using with FreeJoy, you can flash it with st-link or UART.

Yes, you can power controller board from any 5V source (or even 3.3V source)

bayerla, 2020-07-11:
Many thanks. It is Working win 12 keys. But I have problems if  NO key is pressed. Then basis 10k resistor activates Button 1. Would I need 13 / 17 positions  for solution, or it is made by axis config?

Stay health George

vostrenkov, 2020-07-12:
It can be done by tuning ranges of buttons activation. Move first slider a bit to the right to make Button1 activate only if pressed