FreeJoy-Team/FreeJoyConfigurator #18: stm not detected when connected on microusb

victorrun, 2020-07-30:
Hello,

I was able to flash the firmware in the first try, no problems, everything went smoothly just as in the wiki.

Now, when I try to connect the board via micro usb, NOTHING happens (besides the 2 led lights on the board lighting up).
Computer does not detect anything, it's like nothing is plugged, but still the board receives power. What could the problem be? Do you have any clues?

Thank you

vostrenkov, 2020-07-30:
Hello,
If your PC does not report about "device is not recognized" it should be a hardware issue.
First of all make sure that you use verified USB cable (some of them are charging-only). Then check micro-USB socket for bad soldering. If it is ok then check STM32 IC pins soldering.

victorrun, 2020-07-31:
Man, god bless you, it was because of the cable.
I never thought of it because I had tried with two different cables, but seems like both were charging only! Third one was good.

Thank you!

closed 2020-07-31 by vostrenkov