FreeJoy-Team/FreeJoyConfigurator #19: How to deactivate AXES I dont plan to use on my project?

memob777, 2020-08-25:
Hello, 
First I would like to thank you guys for this project, 
It is the best development i have found so far to create HID for games and sims

I am trying to develop a controller for my flight sim.
I just buttons and encoders (About a 100)
and  I got this issues.

- Is there a possibility to deactivate the axes that are shown?  because when i connect my controller board is show 12 axis that I haven configured and don't plan to use on this project.

- Is there a way to accelerate the rate of turn of the encoders? ( i don't know how to better explain than this)
i want to move the Course or heading to 1 degree per click of encoder, but it does 1 degree per 4 clicks, is there a way to multiply the output of the signal so every click is interpreted as 3 or 4?   that will solve all the troubles i have faced so far

Once again thank you very much :D

vostrenkov, 2020-08-25:
Hello,
I appreciate your feedback, glad to know people like this project)

1) There are 2 options that affects on displaying axes and buttons in system: "output" checkbox on corresponding tab (axes or buttons config) and "dynamic hid config" on advanced settings tab.
If "dynamic hid" checked all unused axes and buttons will be hidden for the system, but keep in mind this can break the order of buttons and axes in configurator. I gonna fix it a bit later and make controller be always "dynamic" without checkbox.

2) If you use encoder as axis source "step divider" option will make the requested tuning. Setting lower divider gives you bigger step and vice versa.
But if you use encoders for buttons input you can only select one of three allowed encoder resolutions on encoders tab. You also can combine both techniques but it is not really needed in axis mode as step divider can do all the work.

You may be interested to learn more from the wiki. It is a bit outdated, but most of options work as described

memob777, 2020-08-25:
Thank you very much, 
for issue 1:  - I have followed instructions and removed all extra buttons and switches, except the axis
Is there a way to remove the axis also?  
Issue 2: ill give it a try now :D Thank you very much!! you guys really rock

vostrenkov, 2020-08-25:
[image]

[image]

memob777, 2020-08-25:
Thank you, I just noticed on the windows is as you display.
but on the game is still displaying the axes... might be a glitch from X plane

Thank you for kind and patience :D

memob777, 2020-08-28:
Hello again Vostrenkov
I am wondering if it would be possible to edit the Configuratorto increase the resolution of the "encoder as buttons" to x8 and x16... or maybe it can be done somehow on the code
also for fast turning to increase the rate..

Im trying to do a HDG knob for xplane
I would really appreciate your help

vostrenkov, 2020-08-28:
Hi,
quadrature encoder reading resolution can not be set more than 4x as it only has 4 different state changes.
Some encoders have more than one pulses per detention, maybe using them may help you with the issue

memob777, 2020-08-29:
thank you :D

closed 2020-08-29 by memob777