FreeJoy-Team/FreeJoyConfigurator #24: Issue sharing latches on 4 shift chains

nflaum, 2020-10-03:
Hello,

I'm having an issue configuring the FreeJoy for 128 buttons using 16 shift registers, with a 4 chain x 4 register configuration. I am trying to use a common latch for all 4 chains.

In the Shift Register tab, everything looks fine. I have 4 chains, each chain has 32 buttons, and each latch pin is assigned to B12.

However, there is a problem in the Pin Config tab. On the right side, under Current Config, it only shows 32 total buttons. Upon further examination, it is clear that if I set additional ShiftReg_LATCH pins, I notice the number will increase. However, the board I am using has a shared latch for all 4 chains. In the Button Config tab as well, I only have 32 physical switch to work with, but I should have 128.

I believe this is a bug. The number of physical buttons is tied to the number of pins set to be ShiftReg_LATCH, but if a common latch is used, the number of buttons need to be multiplies by the number of ShiftReg_DATA pins sharing the latch.

Thanks
Noah

vostrenkov, 2020-10-03:
Thanks, it will be fixed soon

vostrenkov, 2020-10-03:
Please check **FreeJoy Configurator v1.6.2b3**

nflaum, 2020-10-07:
> Please check **FreeJoy Configurator v1.6.2b3**

Thank you very much! Works great. Greatly appreciated.

closed 2020-10-07 by nflaum