FreeJoy-Team/FreeJoyConfigurator #25: why can't setup?

liyayo0927, 2020-10-18:
Hello,
When I tried to set up, this problem occurred.
What should I do?

[image]

vostrenkov, 2020-10-18:
You need to install .NET Framework 4.5.2 or newer.
Download link from Microsoft website: https://www.microsoft.com/en-us/download/details.aspx?id=42642

vostrenkov, 2020-10-18:
Are you trying to install using installator from publish/ folder?
You need to use prebuilt .exe from .zip archive on releases page:
https://github.com/FreeJoy-Team/FreeJoyConfigurator/releases/tag/v1.6.2b3

And which operating system do you have?

liyayo0927, 2020-10-18:
I appreciate your message.

_Are you trying to install using installator from publish/ folder?_
Yes. Because I can't found the .exe from FreeJoyConfigurator-master.zip 

_You need to use prebuilt .exe from .zip archive on releases page:
https://github.com/FreeJoy-Team/FreeJoyConfigurator/releases/_
I'm done. And It's working !

And which operating system do you have?
I'm use windos10, .NET Framework it's 4.8.03752

vostrenkov, 2020-10-18:
Nice. I think i should delete this publish folder to prevent such issues

closed 2020-11-18 by vostrenkov