FreeJoy-Team/FreeJoyConfigurator #27: The name in game controller is not change

liyayo0927, 2020-11-01:
Hello,
I have a new issue
Why I change the Device USB name,
but the name in game controller is not change?  
Like this picture.

[image]

vostrenkov, 2020-11-01:
Hi, this is Windows issue it does not update the name from the descriptor.
To get the name changed you need to rename it in Windows registry or delete the instance:

[image]

liyayo0927, 2020-11-01:
Hi, It works !
Thanks a lot for your help !

[image]

closed 2020-11-18 by vostrenkov