FreeJoy-Team/FreeJoyConfigurator #28: Physical pins config and buttons

geologic, 2020-12-12:
Not sure if i understand the relation between pin config and button config...
Lets say at "pin config" tab i set 5 button_ground pins to B12, B13, B14, B15, A8
At "button config" tab i can see on left side my 5 buttons and on the left side the logic attached to them, but how do i know witch pin is physically B12, B13, etc?

I'm asking this because i made a simple test:
I attached one button_ground pin to B12 and it worked as expected, i can see red dot 1 flashing when i push my button.
Later i add 4 button_ground pins to B13, B14, B15, A8 and now my first button on B12 flash the red dot 2 when i press it.
So adding some pins break previous settings.

Can you please clarify?
Thanks

vostrenkov, 2020-12-14:
Hi, buttons enumeration related to pin number. In case of single buttons A0 will get lower number than A1 and B0 will get higher number than A15.
In case of matrix enumeration a bit more complex

geologic, 2020-12-14:
Ah, got it.
Thanks

closed 2020-12-14 by geologic