FreeJoy-Team/FreeJoyConfigurator #29: Possible conflict

evolution808, 2021-01-15:
Strange issue, When using Windows joy.cpl to verify inputs - on opening properties of my controller with a Logitech G923 G HUB steering wheel connected it has this titled at top of window and no inputs of my controller. When Logitech wheel is unplugged I am able to verify inputs on my own controller normally. I have tried changing PID value with same results.

Is anyone able to offer any help?

evolution808, 2021-01-15:
[image]
[image]

vostrenkov, 2021-01-16:
Hi, 
this is possible because of USB HID report architecture. This thing also cause conflicts when using discord/teamspeak and some other apps.
We are working on fix right now

DukeOfDream, 2023-05-20:
Sorry for resurrecting this after more than 2 years but i'm facing the same problem here... I was using a logitech X52 pro in combination with an STM32 FreeJoy based panel and everything was working fine... Today i bought a set of Thrustmaster Rudder Pedals and my device is experiencing the same exact issue... FreeJoy empty page and nothing works... Could it be that the Thrustmaster drivers messed the FreeJoy drivers somehow? My board is a Z690 Apex and i have a 1300w psu and i'm not using any usb hubs... Everything is directly plugged to the motherboard :/ Is there any solution at all?