FreeJoy-Team/FreeJoyConfigurator #30: Setting

speedbird5, 2021-01-22:
In the FreeJoy Wiki here https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Buttons-configuration.md, you say:

_

> (the time while the logical button remains pressed is set in the Toggle Press Time field of the Advanced Settings section);

_

vostrenkov, 2021-01-22:
Hi!
> Do I really need to use diodes
You need them if you suppose your buttons can be pressed simultaneously. You will get ghost pressing without diodes

>  why are the diode directions different in the button matrix for only buttons, but are in the reverse direction when you include an encoder 
If you look closer you will see that both diagrams have cathode to the row pin and anode to the column pin

> But in the Advanced settings section there is no way to set the Toggle Press Time
Wiki is a bit outdated. This settings now available as "Press timer" on Buttons tab. It works now not only for toggle switches but also for other button types

speedbird5, 2021-01-22:
Thanks for the explanations.   I will be using quite a few toggle switches that could have two or more latched on at the same time so diodes will be needed.

Thank you for your work on this project.

vostrenkov, 2021-01-22:
You do "button_nornal" and no tomets applied

speedbird5, 2021-01-22:
> You do "button_nornal" and no timers applied

Ok thanks again

speedbird5, 2021-01-24:
Forgive me as I am a noob but two more dumb questions:
1.  Why do you have two options for wiring switches as 3.3 V or GND ?  Wouldn't you normally just use the GND option and use the Configurator to flip the output ?
2. Is there a way to wire a 4 position rotary switch using the Configurator ?

vostrenkov, 2021-01-24:
1. It is supposed that some people may use freejoy to replace controller on their device. So if you already have your wiring done with external pull-up or pull-down you may choose right option
2. Yes, there are several ways to do this:
- just connect 4 button pins with common gnd/vcc
- make a hardware mode to connect it to analog input and use axis2buttons function

More information can be found [here](https://forums.eagle.ru/topic/190451-rotary-switch-amp-arduino-help-needed)

speedbird5, 2021-01-24:
HI,

> It is supposed that some people may use freejoy to replace controller on their device. So if you already have your wiring done with external pull-up or pull-down you may choose right option
If I am making my own button box with switches, which is the right choice, VCC or GND or doesn’t it matter ?

Thankyou.

vostrenkov, 2021-01-24:
I'd recommend to wire buttons with common GND

speedbird5, 2021-01-24:
Thankyou

speedbird5, 2021-01-24:
Hi,

Is it possible to upload and run a normal Arduino sketch onto the STM 32 after flashing with your firmware and using the configurator ?

Thanks

vostrenkov, 2021-01-25:
Hi, it is possible.
You just need to flash arduino bootloader to stm32

speedbird5, 2021-01-25:
Ok but confirm it will not erase the Freejoy configuration?

vostrenkov, 2021-01-25:
It will for sure

speedbird5, 2021-01-25:
So you’re saying I can’t do both ?  It’s either Freejoy OR Arduino boot loader but not both ?

vostrenkov, 2021-01-25:
I don't know how exactly arduino bootloader works and where it placed in device memory. But freejoy also have its own bootloader in stm32's flash and I'm sure it will be overwritten by arduino bootloader or arduino sketch

speedbird5, 2021-01-25:
Ok thanks for clarifying