FreeJoy-Team/FreeJoyConfigurator #31: Can't Read/Write Config from/to Device

ethan54013, 2021-01-30:
Hi, hope you are doing well.

I have successfully tested and used the Freejoy on my bluepill board before and everything just worked fine.
But recently my FreeJoyConfigurator has been unable to read or write config,
the system log will only be stuck on ''Requesting config..'' or ''Writting config..''
as you can see from the attached picture, and apparently nothing will happen,
[image]
so I would like to ask what could be the problem and how it can be solved?
Thank you for your help in advance!

Reksotiv, 2021-01-30:
Hi. You could try quickly press 2 times Read Config or use https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt

SimProAus, 2021-02-09:
Hi am also having this issue. It is stuck on "Writting" or "Requesting" config. 2 quick presses does nothing, I'm not sure what to do with the link that @Reksotiv gave. I tried this with Two PCs and two Boards

Reksotiv, 2021-02-09:
Link is the new configurator. You can still try to reduce the length of the USB cable

SimProAus, 2021-02-11:
I tried using a shorter USB Cable. Didn't work. How am I supposed to use the "new configurator"? The old one had a .exe the new one doesnt have the same files so I dont know how to install it

Reksotiv, 2021-02-11:
https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt/releases
Download and unpack FreeJoyQt-v1.6.2-1.1.2-Windows.zip